thien94/orb_slam3_ros_wrapper #3: Getting the odom, rather than Pose

rakshith95, 2021-04-27:
Hello @thien94 , 
Thank you for this wrapper. How can I get the `mono_inertial_node` to publish `nav_msgs:Odometry` type, which includes the `twist` information in addition to the `Pose` information?

thien94, 2021-04-28:
Hi @rakshith95 , thank you for the question.

As far as I know, only the pose data are provided by ORB-SLAM3 as it is. After looking into the code, I have a few ideas regarding how to obtain the **linear** velocities. However, how to obtain the **angular** velocities is a mystery to me. Furthermore, the covariance or uncertainty is not computed by ORB-SLAM3 for either pose or velocity. Hence, creating a proper `nav_msgs:Odometry` message is unlikely to be possible unless you figure out the above or accept that the message is incomplete.

Nonetheless, to obtain the velocity my suggestion is:
- Figure out which variable contains the latest velocity and in which frame (global or body), possibly [`mVw`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/include/Frame.h#L196) or [`mVelocity`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/include/Tracking.h#L322), with the former seems more likely.
- Modify [`Tracking.cc`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Tracking.cc) and [`System.cc`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc) to allow access the the aforementioned variables.
- Modify the ROS wrapper (either this package or the provided example) to obtain the velocities, embed them into the ROS message before publishing.

rakshith95, 2021-04-28:
Thank you for your response. I will update here, after attempting the above. Closing this now

closed 2021-04-28 by rakshith95

gjcliff, 2024-10-30:
mVw is linear velocity for the imu:
```cpp
// Frame.h line 185
    // IMU linear velocity
    Eigen::Vector3f mVw;
```