thien94/orb_slam3_ros_wrapper #4: How to run ros bag MH_01_easy.bag

alaamh, 2021-05-10:
I'm getting an error the file not found.

thien94, 2021-05-11:
Can you elaborate on the problem: which file is not found and is that file part of this package?

alaamh, 2021-05-11:
The environment configured as the following:
```
Ubuntu 18.04
ROS Melodic
```
All dependency has been installed.
running following command:
`$ roslaunch orb_slam3_ros_wrapper orb_slam3_mono_inertial_euroc.launch`
gives the below error:
```
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial
[orb_slam3_mono_inertial-2] process has died [pid 2849, exit code -11, cmd /home/alaamh/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial /camera/image_raw:=/cam0/image_raw /imu:=/imu0 __name:=orb_slam3_mono_inertial __log:=/home/alaamh/.ros/log/4d9126dc-b26f-11eb-88e2-000c2902f0b7/orb_slam3_mono_inertial-2.log].
log file: /home/alaamh/.ros/log/4d9126dc-b26f-11eb-88e2-000c2902f0b7/orb_slam3_mono_inertial-2*.log
```
running following command:
`$ rosbag play MH_01_easy.bag`
gives the below error:
```
[ INFO] [1620747730.446810883]: Opening MH_01_easy.bag
[FATAL] [1620747730.530635794]: Error opening file: MH_01_easy.bag
```

thien94, 2021-05-12:
For the first error with `$ roslaunch orb_slam3_ros_wrapper orb_slam3_mono_inertial_euroc.launch`, unfortunately it is not clear what is the problem so you have to investigate a bit further. 
- Firstly, can you run ORB-SLAM3 without ROS (i.e. using the instructions [from the original libary](https://github.com/UZ-SLAMLab/ORB_SLAM3#running-monocular-inertial-node))? If not, then the issue is with the ORB-SLAM3 library.
- Secondly, can you launch the other systems (mono/stereo without IMU, stereo with IMU) normally?
- Additionally, you can take a look at the log file (shown in the error message) to obtain more info.

For the second error with `$ rosbag play MH_01_easy.bag`: have you navigate to the folder where the bag file is located? You can only run that command when you provide the correct path to the bag file or are in the same folder.

alaamh, 2021-05-16:
I could not figure it out, the mentioned log file orb_slam3_mono_inertial-2 does not exist and other logos for ROS have no errors, what is the easy way to test ORB_SLAM3 itself? Please note it has built without any error and all dependency library is installed successfully.

alaamh, 2021-05-20:
it seems I have missed getting EuRoc Dataset:
`wget -c http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/machine_hall/MH_01_easy/MH_01_easy.zip`

then Unzip to MH01 folder :)

thien94, 2021-05-26:
Hi @alaamh is the issue resolved? If yes, you can close the issue for now. Feel free to reopen it if the need arises.

closed 2021-05-30 by alaamh