thien94/orb_slam3_ros_wrapper #6: [orb_slam3_rbgd-1] process has died

siddharthcb, 2021-06-16:
hi,
i have built ORB_SLAM3 which works well directly using `rosrun ORB_SLAM3 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/d455.yaml`
i have also followed all your steps and built this package from source. when i launch` roslaunch orb_slam3_ros_wrapper orb_slam3_rgbd_tum.launch`  I get an error and unable to visualize on rviz.

```
$  roslaunch orb_slam3_ros_wrapper orb_slam3_rgbd_tum.launch

logging to /home/ap/.ros/log/b43cc708-ce60-11eb-93f7-d037455179ee/roslaunch-home-28724.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/ap/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.8.226:41125/

SUMMARY
========

PARAMETERS
 * /orb_slam3_rbgd/map_frame_id: world
 * /orb_slam3_rbgd/pose_frame_id: camera_link
 * /orb_slam3_rbgd/settings_file: /home/ap/catkin_w...
 * /orb_slam3_rbgd/voc_file: /home/ap/catkin_w...
 * /orb_slam3_ros/trajectory_server_orb_slam3/source_frame_name: /camera_link
 * /orb_slam3_ros/trajectory_server_orb_slam3/target_frame_name: /world
 * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_publish_rate: 20.0
 * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_update_rate: 20.0
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    orb_slam3_rbgd (orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_rgbd)
    rviz (rviz/rviz)
  /orb_slam3_ros/
    trajectory_server_orb_slam3 (hector_trajectory_server/hector_trajectory_server)

ROS_MASTER_URI=http://192.168.8.226:11311

process[orb_slam3_rbgd-1]: started with pid [28739]
process[rviz-2]: started with pid [28740]
process[orb_slam3_ros/trajectory_server_orb_slam3-3]: started with pid [28741]
[ INFO] [1623830740.230521072]: Waiting for tf transform data between frames /world and /camera_link to become available
[ INFO] [1623830740.637291368]: rviz version 1.13.15
[ INFO] [1623830740.637436474]: compiled against Qt version 5.9.5
[ INFO] [1623830740.637483610]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1623830740.653172300]: Forcing OpenGl version 0.

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D
[orb_slam3_rbgd-1] process has died [pid 28739, exit code -11, cmd /home/ap/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_rgbd /camera/rgb/image_raw:=/camera/color/image_raw /camera/depth_registered/image_raw:=/camera/aligned_depth_to_color/image_raw __name:=orb_slam3_rbgd __log:=/home/ap/.ros/log/b43cc708-ce60-11eb-93f7-d037455179ee/orb_slam3_rbgd-1.log].
log file: /home/ap/.ros/log/b43cc708-ce60-11eb-93f7-d037455179ee/orb_slam3_rbgd-1*.log
[ INFO] [1623830741.651802965]: Stereo is NOT SUPPORTED
[ INFO] [1623830741.652021249]: OpenGl version: 4.6 (GLSL 4.6).
[ WARN] [1623830760.235955973]: No transform between frames /world and /camera_link available after 20.005462 seconds of waiting. This warning only prints once.
```
orb_slam3_rgbd_tum.launch
```
<launch>
    <node name="orb_slam3_rbgd" pkg="orb_slam3_ros_wrapper" type="orb_slam3_ros_wrapper_rgbd" output="screen">
        <!-- for EuRoC dataset -->
        <remap from="/camera/rgb/image_raw"                 to="/camera/color/image_raw"/>
        <remap from="/camera/depth_registered/image_raw"    to="/camera/aligned_depth_to_color/image_raw"/>

        <!-- Parameters for original ORB-SLAM3 -->
        <param name="voc_file"      type="string"   value="$(find orb_slam3_ros_wrapper)/config/ORBvoc.txt" />
        <param name="settings_file" type="string"   value="$(find orb_slam3_ros_wrapper)/config/d455.yaml" />
    
        <!-- Parameters for ROS -->
        <param name="map_frame_id"  type="string"   value="world" />
        <param name="pose_frame_id" type="string"   value="camera_link" />
    </node>
    
    <!-- Visualization - RViz-->
    <node name="rviz" pkg="rviz" type="rviz" args="-d $(find orb_slam3_ros_wrapper)/config/orb_slam3.rviz" output="screen" />

    <node pkg="hector_trajectory_server" type="hector_trajectory_server" name="trajectory_server_orb_slam3" output="screen" ns="orb_slam3_ros" >
        <param name="/target_frame_name" value="/world" />
        <param name="/source_frame_name" value="/camera_link" />
        <param name="/trajectory_update_rate" value="20.0" />
        <param name="/trajectory_publish_rate" value="20.0" />
    </node>
</launch>
```
 I am using Melodic and i have confirmed that i have  all the dependencies installed. What am i doing wrong?

AnhTuDo1998, 2021-06-25:
Hi, have you check the voc_file and setting_file exists under the stated path? If you haven't extract the vocabulary file, this might be an issue that can crash the node.

jfenton888, 2022-02-15:
@Abner0907 this error actually comes from rviz surprisingly. I am actually getting the exact same error message when trying to run a monocular setup. Commenting out the `rviz` line in `orb_slam3_mono_euroc.launch` causes that info message to disappear but I still get the exact same error message as @siddharthcb and cannot figure out why

jfenton888, 2022-02-15:
I just "solved" this issue. It seems to be something to do with some changes that were made to ORB_SLAM3 when going from `v0.3` to `v0.4`, and checking out to `v0.3` allows whatever issue is going on to be resolved. I would definitely appreciate if someone could look into this so I could use any of the changes made in this update but this seems to work for now

RoZhong, 2022-03-07:
> I just "solved" this issue. It seems to be something to do with some changes that were made to ORB_SLAM3 when going from `v0.3` to `v0.4`, and checking out to `v0.3` allows whatever issue is going [...]
Is it possible to solve this problem by using v0.3 version?

AlanMohammad20, 2022-03-17:
#siddharthcb 
hello sir can you please tell me how you run the orbslam3 algorithm
I have build the orbslam3 successfully but i steel doesn't know how to run the algorithm
I have installed the intel realsense SDK and ROS wrapper
P.S I am using DEPTH CAMERA D455

TheMotonaut, 2022-07-28:
So this seems to be still a problem. Downgrading to v0.3 removes the "exit code -11" error but instead gives me undefined symbol error and "exit code 127". Anyone got any update?

TheMotonaut, 2022-07-29:
So the problem existed in the fact that the node was init with a different name compared to the launch file node name. So the program fails to load the settings files and crashes.

thien94, 2022-08-15:
Hi, sorry for the late reply.

I have updated the code to work with V1.0 of ORB-SLAM3 as well as some other changes. If you are still interested, please test it out and let me know if the problem is still there.

Smark2019, 2022-08-18:
> Hi, sorry for the late reply.
> 
> I have updated the code to work with V1.0 of ORB-SLAM3 as well as some other changes. If you are still interested, please test it out and let me know if the problem is still there.

Hi,
I am actually having same problem (I am using D435i btw as camera), although I have just updated the repo from GitHub after your changes; there is still same error. I have launch 2 launch file as given following:
`pehlivanoglu@pehlivanoglu:~$ roslaunch orb_slam3_ros_wrapper  orb_slam3_tum_rgbd.launch 
`

> ... logging to /home/pehlivanoglu/.ros/log/12a70a60-1edd-11ed-834a-55711dcc030a/roslaunch-pehlivanoglu-12208.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://127.0.0.1:36013/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /orb_slam3_rbgd/map_frame_id: world
>  * /orb_slam3_rbgd/pose_frame_id: camera
>  * /orb_slam3_rbgd/settings_file: /home/pehlivanogl...
>  * /orb_slam3_rbgd/voc_file: /home/pehlivanogl...
>  * /orb_slam3_ros/trajectory_server_orb_slam3/source_frame_name: /camera
>  * /orb_slam3_ros/trajectory_server_orb_slam3/target_frame_name: /world
>  * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_publish_rate: 20.0
>  * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_update_rate: 20.0
>  * /rosdistro: noetic
>  * /rosversion: 1.15.14
> 
> NODES
>   /
>     orb_slam3_rbgd (orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_rgbd)
>     rviz (rviz/rviz)
>   /orb_slam3_ros/
>     trajectory_server_orb_slam3 (hector_trajectory_server/hector_trajectory_server)
> 
> auto-starting new master
> process[master]: started with pid [12227]
> ROS_MASTER_URI=http://127.0.0.1:11311/
> 
> setting /run_id to 12a70a60-1edd-11ed-834a-55711dcc030a
> process[rosout-1]: started with pid [12248]
> started core service [/rosout]
> process[orb_slam3_rbgd-2]: started with pid [12253]
> process[rviz-3]: started with pid [12256]
> process[orb_slam3_ros/trajectory_server_orb_slam3-4]: started with pid [12257]
> [ INFO] [1660817037.415718561]: Waiting for tf transform data between frames /world and /camera to become available
> 
> ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> This program comes with ABSOLUTELY NO WARRANTY;
> This is free software, and you are welcome to redistribute it
> under certain conditions. See LICENSE.txt.
> 
> Input sensor was set to: RGB-D
> Loading settings from /home/pehlivanoglu/orb_ws/src/orb_slam3_ros_wrapper/config/TUM1.yaml
> -Loaded camera 1
> Camera.newHeight optional parameter does not exist...
> Camera.newWidth optional parameter does not exist...
> -Loaded image info
> -Loaded RGB-D calibration
> -Loaded ORB settings
> Viewer.imageViewScale optional parameter does not exist...
> -Loaded viewer settings
> System.LoadAtlasFromFile optional parameter does not exist...
> System.SaveAtlasToFile optional parameter does not exist...
> -Loaded Atlas settings
> System.thFarPoints optional parameter does not exist...
> -Loaded misc parameters
> ----------------------------------
> SLAM settings: 
> -Camera 1 parameters (Pinhole): [ 517.306 516.469 318.643 255.314 ]
> -Camera 1 distortion parameters: [  0.262383 -0.953104 -0.005358 0.002628 1.16331 ]
> -Original image size: [ 640 , 480 ]
> -Current image size: [ 640 , 480 ]
> -Sequence FPS: 30
> -RGB-D depth map factor: 5000
> -Features per image: 1000
> -ORB scale factor: 1.2
> -ORB number of scales: 8
> -Initial FAST threshold: 20
> -Min FAST threshold: 7
> 
> 
> Loading ORB Vocabulary. This could take a while...
> Vocabulary loading failure: This is not a correct text file!
> Wrong path to vocabulary. 
> Falied to open at: /home/pehlivanoglu/orb_ws/src/orb_slam3_ros_wrapper/config/ORBvoc.txt
> [ INFO] [1660817037.581468112]: rviz version 1.14.14
> [ INFO] [1660817037.581548110]: compiled against Qt version 5.12.8
> [ INFO] [1660817037.581571975]: compiled against OGRE version 1.9.0 (Ghadamon)
> [ INFO] [1660817037.594919125]: Forcing OpenGl version 0.
> [orb_slam3_rbgd-2] process has died [pid 12253, exit code 255, cmd /home/pehlivanoglu/orb_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_rgbd /camera/rgb/image_raw:=/camera/rgb/image_color /camera/depth_registered/image_raw:=/camera/depth/image __name:=orb_slam3_rbgd __log:=/home/pehlivanoglu/.ros/log/12a70a60-1edd-11ed-834a-55711dcc030a/orb_slam3_rbgd-2.log].
> log file: /home/pehlivanoglu/.ros/log/12a70a60-1edd-11ed-834a-55711dcc030a/orb_slam3_rbgd-2*.log
> [ INFO] [1660817037.845290668]: Stereo is NOT SUPPORTED
> [ INFO] [1660817037.845356631]: OpenGL device: AMD RENOIR (DRM 3.42.0, 5.15.0-46-generic, LLVM 12.0.0)
> [ INFO] [1660817037.845374263]: OpenGl version: 4,6 (GLSL 4,6) limited to GLSL 1.4 on Mesa system.
> [ WARN] [1660817057.419467386]: No transform between frames /world and /camera available after 20.003783 seconds of 
> 
> 

and also at another terminal I launched:

`pehlivanoglu@pehlivanoglu:~$ roslaunch realsense2_camera rs_camera.launch `

But on Rviz, there is nothing to show. Please help about it. Im waiting your suggestions to that.

thien94, 2022-08-18:
Hi @Smark2019 ,

In your case, the issue is that no `ORBvoc.txt` file was found.

You can copy the `ORBvoc.txt` file from `ORB-SLAM3/Vocabulary/` folder to the `config` folder in this package. Alternatively, you can change the `voc_file` param in the launch file to point to the right folder.

I should include this instruction in the README. Thank you for letting me know.

Smark2019, 2022-08-18:
Hi @thien94 ;

Firstly thanks for your fast reply, I did what you suggested to me and now I am able to see Map Viewer GUI and and Camera Viewer GUI showed by ORB SLAM but unfortunately without any camera image frame. What I do now is stated at bottom respectively:

I launch first: `pehlivanoglu@pehlivanoglu:~$ roslaunch realsense2_camera rs_camera.launch` (Btw, it is supposed to be run, right?)

 I also should inform you that I have changed a line at the `orb_slam3_euroc_mono.launch` file. Because I think that topic name of `remap from ` should be equal to topic of `/camera/color/image_raw` generated by Intel Realsense ROS wrapper that I launched first. So, I decided to change the name of it to equalize. I am asking because maybe I am wrong.

> 
>         <remap from="/camera/color/image_raw" to="/cam0/image_raw"/>
> 

After that I launched: `pehlivanoglu@pehlivanoglu:~$ roslaunch orb_slam3_ros_wrapper orb_slam3_tum_rgbd.launch`

But I am getting the following as output on the terminal of above command line:

> [ WARN] [1660826198.610959881]: No transform between frames /world and /camera available after 20.002947 seconds of waiting. This warning only prints once.

thien94, 2022-08-19:
Hi @Smark2019, I think your remapping param might be the issue. 
> I have changed a line at the `orb_slam3_euroc_mono.launch` file. 

If you launch `orb_slam3_tum_rgbd.launch` then shouldn't the changes be made in that same rgbd file instead?

> I think that topic name of remap **from** should be equal to topic of /camera/color/image_raw

The part that you need to change is the **to** part, not the **from**.

Smark2019, 2022-08-19:
> Hi @Smark2019, I think your remapping param might be the issue.
>
>
> If you launch `orb_slam3_tum_rgbd.launch` then shouldn't the changes be made in that same rgbd file instead?
>
>
> The part that you [...]

Ohw, it should be mistakenly written there. That's my bad.Sorry for that. Yes, same file should be changed sure. So, I will try to do your suggesstions regarding remapping parameters and let you know about it. Thanks a lot.

vinsco30, 2022-08-23:
Hi @thien94, 
I am trying to run examples from EuRoC dataset following the instructions that you wrote in the README file. Every time I launch orb slam 3 with the following command line `roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_monoimu.launch` it opens rviz without problems then it tries to open the viewer, the window appears and after few moments it disappears with the message on the terminal that the process has died. I've followed your instructions for the installation and the version of ORB SLAM3 works without problems. Any suggestion on how can I fix this problem? I attach the messages on the terminal:

> vinsco@vinsco:~$ roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_monoimu.launch 
... logging to /home/vinsco/.ros/log/3cd13bb2-2301-11ed-9772-1d1433e69a22/roslaunch-vinsco-16102.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vinsco:36737/

SUMMARY
========

PARAMETERS
 * /orb_slam3_mono_inertial/map_frame_id: world
 * /orb_slam3_mono_inertial/pose_frame_id: camera
 * /orb_slam3_mono_inertial/settings_file: /home/vinsco/catk...
 * /orb_slam3_mono_inertial/voc_file: /home/vinsco/catk...
 * /orb_slam3_ros/trajectory_server_orb_slam3/source_frame_name: /camera
 * /orb_slam3_ros/trajectory_server_orb_slam3/target_frame_name: /world
 * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_publish_rate: 20.0
 * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_update_rate: 20.0
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    orb_slam3_mono_inertial (orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial)
    rviz (rviz/rviz)
  /orb_slam3_ros/
    trajectory_server_orb_slam3 (hector_trajectory_server/hector_trajectory_server)

auto-starting new master
process[master]: started with pid [16117]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3cd13bb2-2301-11ed-9772-1d1433e69a22
process[rosout-1]: started with pid [16134]
started core service [/rosout]
process[orb_slam3_mono_inertial-2]: started with pid [16141]
process[rviz-3]: started with pid [16142]
process[orb_slam3_ros/trajectory_server_orb_slam3-4]: started with pid [16143]
[ INFO] [1661272374.719985319]: Waiting for tf transform data between frames /world and /camera to become available

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial
Loading settings from /home/vinsco/catkin_ws/src/orb_slam3_ros_wrapper/config/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
IMU.InsertKFsWhenLost optional parameter does not exist...
	-Loaded IMU calibration
	-Loaded ORB settings
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 458.654 457.296 367.215 248.375 ]
	-Camera 1 distortion parameters: [  -0.283408 0.0739591 0.00019359 1.76187e-05 ]
	-Original image size: [ 752 , 480 ]
	-Current image size: [ 752 , 480 ]
	-Sequence FPS: 20
	-Gyro noise: 0.00017
	-Accelerometer noise: 0.002
	-Gyro walk: 1.9393e-05
	-Accelerometer walk: 0.003
	-IMU frequency: 200
	-Features per image: 1200
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
[ INFO] [1661272374.835826868]: rviz version 1.14.14
[ INFO] [1661272374.835865764]: compiled against Qt version 5.12.8
[ INFO] [1661272374.835882687]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1661272374.844413134]: Forcing OpenGl version 0.
[ INFO] [1661272375.028629895]: Stereo is NOT SUPPORTED
[ INFO] [1661272375.028680989]: OpenGL device: Mesa Intel(R) UHD Graphics 620 (WHL GT2)
[ INFO] [1661272375.028697356]: OpenGl version: 4,6 (GLSL 4,6) limited to GLSL 1.4 on Mesa system.
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
[orb_slam3_mono_inertial-2] process has died [pid 16141, exit code -11, cmd /home/vinsco/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial /camera/image_raw:=/cam0/image_raw /imu:=/imu0 __name:=orb_slam3_mono_inertial __log:=/home/vinsco/.ros/log/3cd13bb2-2301-11ed-9772-1d1433e69a22/orb_slam3_mono_inertial-2.log].
log file: /home/vinsco/.ros/log/3cd13bb2-2301-11ed-9772-1d1433e69a22/orb_slam3_mono_inertial-2*.log
^C[orb_slam3_ros/trajectory_server_orb_slam3-4] killing on exit
[rviz-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

thien94, 2022-08-24:
Hi @vinsco30, from the terminal output it is not clear to me what might be the problem. 
- Are you using the latest version of ORB-SLAM3 (V1.0)? This repo might not work with previous versions anymore.
- Can you look into the log file `log file: /home/vinsco/.ros/log/3cd13bb2-2301-11ed-9772-1d1433e69a22/orb_slam3_mono_inertial-2*.log` for any error messages?

vinsco30, 2022-08-24:
Thank you for the quick answer @th

> Hi @vinsco30, from the terminal output it is not clear to me what might be the problem.
>
> * Are you using the latest version of ORB-SLAM3 (V1.0)? This repo might not work with previous versions [...]

Hi, yes I am using ORB-SLAM3 (V1.0). About the log file it is impossible to understand something (it's very huge...)

Smark2019, 2022-08-25:
Hi again @thien94, After applying your suggestions to me; now I can open viewer and rviz but still getting error which is different that previous. I am sharing the output of the terminal at the bottom for your consider:

> pehlivanoglu@pehlivanoglu:~/ORB_SLAM3$ roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_stereoimu.launch 
> ... logging to /home/pehlivanoglu/.ros/log/546e6934-2482-11ed-a809-a16991ff33c2/roslaunch-pehlivanoglu-42066.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://127.0.0.1:45573/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /orb_slam3_ros/trajectory_server_orb_slam3/source_frame_name: /camera
>  * /orb_slam3_ros/trajectory_server_orb_slam3/target_frame_name: /world
>  * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_publish_rate: 20.0
>  * /orb_slam3_ros/trajectory_server_orb_slam3/trajectory_update_rate: 20.0
>  * /orb_slam3_stereo_inertial/map_frame_id: world
>  * /orb_slam3_stereo_inertial/pose_frame_id: camera
>  * /orb_slam3_stereo_inertial/settings_file: /home/pehlivanogl...
>  * /orb_slam3_stereo_inertial/voc_file: /home/pehlivanogl...
>  * /rosdistro: noetic
>  * /rosversion: 1.15.14
> 
> NODES
>   /
>     orb_slam3_stereo_inertial (orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_stereo_inertial)
>     rviz (rviz/rviz)
>   /orb_slam3_ros/
>     trajectory_server_orb_slam3 (hector_trajectory_server/hector_trajectory_server)
> 
> auto-starting new master
> process[master]: started with pid [42085]
> ROS_MASTER_URI=http://127.0.0.1:11311/
> 
> setting /run_id to 546e6934-2482-11ed-a809-a16991ff33c2
> process[rosout-1]: started with pid [42106]
> started core service [/rosout]
> process[orb_slam3_stereo_inertial-2]: started with pid [42113]
> process[rviz-3]: started with pid [42114]
> process[orb_slam3_ros/trajectory_server_orb_slam3-4]: started with pid [42115]
> [ INFO] [1661437770.764162644]: Waiting for tf transform data between frames /world and /camera to become available
> 
> ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> This program comes with ABSOLUTELY NO WARRANTY;
> This is free software, and you are welcome to redistribute it
> under certain conditions. See LICENSE.txt.
> 
> Input sensor was set to: Stereo-Inertial
> Loading settings from /home/pehlivanoglu/orb_ws/src/orb_slam3_ros_wrapper/config/EuRoC.yaml
> Camera1.k3 optional parameter does not exist...
> -Loaded camera 1
> Camera2.k3 optional parameter does not exist...
> -Loaded camera 2
> Camera.newHeight optional parameter does not exist...
> Camera.newWidth optional parameter does not exist...
> -Loaded image info
> IMU.InsertKFsWhenLost optional parameter does not exist...
> -Loaded IMU calibration
> -Loaded ORB settings
> -Loaded viewer settings
> System.LoadAtlasFromFile optional parameter does not exist...
> System.SaveAtlasToFile optional parameter does not exist...
> -Loaded Atlas settings
> System.thFarPoints optional parameter does not exist...
> -Loaded misc parameters
> -Computed rectification maps
> ----------------------------------
[... 50 lines cut ...]
>         
>         <remap from="/camera/image_raw" to="/camera/color/image_raw"/>
>         <remap from="/camera/depth_registered/image_raw" to="/camera/depth/image_rect_raw"/>
> 
>         
>         <param name="voc_file"      type="string"   value="$(find orb_slam3_ros_wrapper)/config/ORBvoc.txt" />
>         <param name="settings_file" type="string"   value="$(find orb_slam3_ros_wrapper)/config/TUM1.yaml" />
>     
>         
>         <param name="map_frame_id"  type="string"   value="world" />
>         <param name="pose_frame_id" type="string"   value="camera" />
>     </node>
>     
>     
>     <node name="rviz" pkg="rviz" type="rviz" args="-d $(find orb_slam3_ros_wrapper)/config/orb_slam3.rviz" output="screen" />
> 
>     <node pkg="hector_trajectory_server" type="hector_trajectory_server" name="trajectory_server_orb_slam3" output="screen" ns="orb_slam3_ros" >
>         <param name="/target_frame_name" value="/world" />
>         <param name="/source_frame_name" value="/camera" />
>         <param name="/trajectory_update_rate" value="20.0" />
>         <param name="/trajectory_publish_rate" value="20.0" />
>     </node>
> </launch>

Thanks for your interest.

thien94, 2022-08-26:
Hi @Smark2019 , you can ignore the message `No transform between frames /world and /camera available` as it is only a warning from `hector_trajectory_server`. It is not related to either the wrapper or ORB-SLAM3.

So after you launch `roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_stereoimu.launch`, did you play the bag file (`rosbag play MH_01_easy.bag`) and were there any other errors?

vigneshwarvenkat, 2022-11-02:
Hi, I had the same issues when trying to run the EUROC dataset. Here are some things I noticed after launching 

1. When checking the rqt_graph, the orb_slam3_mono_inertial node seems to be dead as well.
[image]

2. When using rviz, there seems to be no tf data.
 
[image]

It worked after changing the <roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_monoimu.launch> to <roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_mono.launch>

thien94, 2022-11-02:
Hi @vigneshwarvenkat , can you check on the console output for any error messages (usually in red)? Additionally, can you run the stereo-inertial launch file?

mgrallos, 2023-06-05:
> exit code 127

I had the same problem with exit code 127 I learn it was because of multiple installations and versions of DBoW2. You can remove all previous and related files in /usr/local/

mgrallos, 2023-06-05:
> Hi @thien94 , I have same issue with @vinsco30 and @vigneshwarvenkat , euroc_mono.launch has no problem, but euroc_monoimu.launch, euroc_stereo.launch, and euroc_stereoimu.launch will all die with exit code -11.

as for the exit code 11, some say it is due to the pangolin driver. I tried setting the parameter "enable pangolin" to false and it remove the error message but it occurs when I tried to run a dataset

Nico-Sander, 2024-07-16:
Has anyone found a solution to the "exit code -11 / process has died" problem?

anmar27, 2025-03-06:
> Has anyone found a solution to the "exit code -11 / process has died" problem?

I was able to solve the issue by downgrading the opencv version in the CMakelist from 4.4 to 3.2.