thien94/orb_slam3_ros_wrapper #7: Could publish all point clouds of map

Hyper-Devil, 2021-06-16:
Thank you for your outstanding work.

I am doing some SLAM-related work. The current point cloud topic _/orb_slam3_ros/map_points_ published currently only contains the points currently tracked. If we can publish all the point clouds of the main map, It will be easy to create an occupancy grid map in ROS,  I believe orb-slam3 will be easier to use. Do you have an update plan?

Thanks again for your work.

palakons, 2021-07-10:
I am doing the same thing for my class project. My plan though is to let [RTAB-Map](http://wiki.ros.org/rtabmap_ros) consuming the PointCloud2 output from ORB_SLAM3, and viola! 3D occupancy map.

What do you think?

thien94, 2021-07-28:
Hi @Hyper-Devil , so sorry for the late reply.

At the moment, it seems that to access the point cloud of the whole map, we need to use this function [`GetAllMapPoints`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/093173baccc70930c87c64bce5b54e5664d6535f/src/Atlas.cc#L157-L161). The trouble is that there is no API that exposes the function directly to our ROS wrapper so some modifications are required. One solution is:

1. modify `System` class to have the `get` function that returns the whole map's point cloud obtained from `mpAtlas`.
2. modify the wrapper accordingly to inquire and receive the data, then convert it to ROS-friendly format ([`PointCloud`](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/PointCloud.html))
3. publish the point cloud msg

Since it requires noticeable changes with the ORB-SLAM3 library itself, it may not be available anytime soon (there is a backlog of PRs with ORB-SLAM3). You can go with the solution as @palakons suggested which is much more straightforward.

Hyper-Devil, 2021-11-02:
Thank you for your advice and thank you again for your work.

closed 2021-11-02 by Hyper-Devil

hitungtung, 2022-06-29:
> Thank you for your advice and thank you again for your work.

Did you solve the problem?