thien94/orb_slam3_ros_wrapper #8: Stops every time there is no motion, and restarts odometry tracking.

rakshith95, 2021-07-13:
Hello, 
Thank you again for your work on the wrapper. 
While running with a rosbag of a drone flight, I find that whenever there is no motion, like when it hovers, the ORB slam stops tracking and re-initializes. Any idea why this could be happening?

thien94, 2021-07-14:
I think it might be related to this issue https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/172
You can plot out the odometry frame in the viewer to confirm this as well. If the odometry resets to zeros, it's likely to be the same issue.

rakshith95, 2021-07-14:
It does reset to zero each time, which makes it virtually unusable.

Do you have a recommendation for any other Mono VIO system which works well on UAVs in your experience?

thien94, 2021-07-14:
[VINS-Fusion](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion) and [OpenVins](https://github.com/rpng/open_vins) are two of the most robust and well supported in my opinion. They are also the representatives of different schools of thought (optimization-based and EKF-based) to the VIO as well.

rakshith95, 2021-07-14:
Thanks for the suggestions. I find that OpenVins is too parameter reliant for my liking, but trying that out now anyway.

closed 2021-07-14 by rakshith95