thien94/orb_slam3_ros_wrapper #10: Uncertainty/covariance publisher node?

maghauke, 2021-11-02:
Hi, do you know if it is possible to make a node that publish the covariance/uncertainty matrix?

thien94, 2021-11-03:
If you can obtain reliable pose uncertainty, you can modify `pose_pub` to publish the more general [geometry_msgs/PoseWithCovarianceStamped](http://docs.ros.org/en/noetic/api/geometry_msgs/html/msg/PoseWithCovarianceStamped.html) message instead of `PoseStamped`.

But the bigger question is how do you obtain the pose uncertainty, since afaik ORB-SLAM3 does not provide one https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/105

maghauke, 2021-11-03:
I thought that you could use a hessian approximation but I do not have enough insight into the orbslam3 code…

closed 2022-08-15 by thien94