thien94/orb_slam3_ros_wrapper #11: Building errors

danydownload, 2022-03-02:
Hi,

I cannot manage to build this ros package.
I got the following errors: 

[image]
[image]
[image]

Do you have any suggest?

Libraries version:
Ros Noetic
OPENCV VERSION: 4.5.5
Eigen 3 VERSION: 3.3.7
and the latest version of Sophus

RoZhong, 2022-03-07:
try c++11 to c++14

danydownload, 2022-03-10:
already tried but no luck!

mikodham, 2022-03-29:
> already tried but no luck!

same here!

cmanore25, 2022-04-06:
I made a fork of this repo located [here](https://github.com/cmanore25/orb_slam3_ros_wrapper).

After a lot of source code changes, I was able to test it with the EuRoC stereo inertial launch file. I set the OpenCV required version to 4.2 (change it in the CMakeList if you need to). Also change the orbslam path in the CMakeList, per the readme instructions. You'll also need to point to the new vocabulary and yaml files from the original orbslam package.

Biggest source code changes were including the Converter.h header file and implementing the conversion to cv::Mat format, and correctly indexing the Vector3f values in common.cc.

koksyuen, 2022-06-01:
@cmanore25 which version of ROS (Melodic/Noetic) are you using?

cmanore25, 2022-06-01:
> @cmanore25 which version of ROS (Melodic/Noetic) are you using?

I am using ROS Noetic with Ubuntu 20.04

viveksood97, 2022-06-06:
Did anyone find a solution for this issue with the latest ORB_SLAM3 version?

viveksood97, 2022-06-06:
> I made a fork of this repo located [here](https://github.com/cmanore25/orb_slam3_ros_wrapper).
>
> After a lot of source code changes, I was able to test it with the EuRoC stereo inertial launch file. [...]

Hey I tried to use your fork but still got the same error. Any suggestions?

Pipiaaaa, 2022-06-17:
@cmanore25  
hi, I checked your fork of this repo and errors above are solved by using your method. Thanks a lot.
However, I still have another error.
Do  you have any idea to solve the following problem?
[image]

Srijal97, 2022-08-08:
@cmanore25 Thank you, I was able to run it on Ubuntu 20 as well. I had sophus header file not found errors during build which I circumvented by changing all instances of `#include "sophus/*.hpp"` to `#include "Thirdparty/Sophus/sophus/*.hpp"` in `ORB_SLAM3/include` before building it. 

Also I had to add `export LD_LIBRARY_PATH=/root/ORB_SLAM3/lib:/root/ORB_SLAM3/Thirdparty/g2o/lib:/root/ORB_SLAM3/Thirdparty/DBoW2/lib:$LD_LIBRARY_PATH` to `.bashrc` to get rid of symbol not found errors.

thien94, 2022-08-15:
Hi @Srijal97 and @viveksood97 and others,

Sorry for the late reply and update. I have updated the repo to work with ORB-SLAM3 V1.0 and Ubuntu 20.04. There are also other changes to remove unnecessary stuffs.

If you are still interested, please use the latest version and let me know if the problem persists.

Thank you and hope this helps.

zoldaten, 2022-11-16:
i failed with building on Ubuntu 20,04 odroid xu4:
```
[  5%] Linking CXX executable /home/odroid/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 3 (>= sh_info of 3)
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 4 (>= sh_info of 3)
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 5 (>= sh_info of 3)
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 6 (>= sh_info of 3)
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 7 (>= sh_info of 3)
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 8 (>= sh_info of 3)
/usr/bin/ld: /usr/lib/arm-linux-gnueabihf/libEGL.so: .dynsym local symbol at index 9 (>= sh_info of 3)
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `unsigned long long fmt::v6::internal::width_checker<fmt::v6::internal::error_handler>::operator()<bool, 0>(bool)':
/usr/include/fmt/format.h:1977: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `unsigned long long fmt::v6::internal::precision_checker<fmt::v6::internal::error_handler>::operator()<bool, 0>(bool)':
/usr/include/fmt/format.h:1997: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `void fmt::v6::print<Sophus::SO3Base<Sophus::SO3<float, 0> >::normalize()::{lambda()#1}::operator()() const::str, Eigen::Transpose<Sophus::SO3Base<Sophus::SO3<float, 0> >::normalize()::{lambda()#1}::operator()() const::str::Matrix<float, 4, 1, 0, 4, 1> >, 0>(Sophus::SO3Base<Sophus::SO3<float, 0> >::normalize()::{lambda()#1}::operator()() const::str const&, Eigen::Transpose<Sophus::SO3Base<Sophus::SO3<float, 0> >::normalize()::{lambda()#1}::operator()() const::str::Matrix<float, 4, 1, 0, 4, 1> >&&)':
/usr/include/fmt/core.h:1514: undefined reference to `fmt::v6::vprint(fmt::v6::basic_string_view<char>, fmt::v6::format_args)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `void fmt::v6::print<char [2], , 0>(char const (&) [2])':
/usr/include/fmt/core.h:1514: undefined reference to `fmt::v6::vprint(fmt::v6::basic_string_view<char>, fmt::v6::format_args)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `std::make_unsigned<int>::type fmt::v6::internal::to_unsigned<int>(int)':
/usr/include/fmt/core.h:266: undefined reference to `fmt::v6::internal::assert_fail(char const*, int, char const*)'
/usr/bin/ld: /usr/include/fmt/core.h:266: undefined reference to `fmt::v6::internal::assert_fail(char const*, int, char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char>::on_error(char const*)':
/usr/include/fmt/core.h:1169: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `int fmt::v6::internal::get_dynamic_spec<fmt::v6::internal::width_checker, fmt::v6::basic_format_arg<fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char> >, fmt::v6::internal::error_handler>(fmt::v6::basic_format_arg<fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char> >, fmt::v6::internal::error_handler)':
/usr/include/fmt/format.h:2118: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `unsigned long long fmt::v6::internal::width_checker<fmt::v6::internal::error_handler>::operator()<int, 0>(int)':
/usr/include/fmt/format.h:1971: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `int fmt::v6::internal::get_dynamic_spec<fmt::v6::internal::width_checker, fmt::v6::basic_format_arg<fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char> >, fmt::v6::internal::error_handler>(fmt::v6::basic_format_arg<fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char> >, fmt::v6::internal::error_handler)':
/usr/include/fmt/format.h:2118: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `unsigned long long fmt::v6::internal::width_checker<fmt::v6::internal::error_handler>::operator()<int, 0>(int)':
/usr/include/fmt/format.h:1971: undefined reference to `fmt::v6::internal::error_handler::on_error(char const*)'
/usr/bin/ld: CMakeFiles/orb_slam3_ros_wrapper_mono.dir/src/mono_node.cc.o: in function `decltype ({parm#1}(0)) fmt::v6::visit_format_arg<fmt::v6::internal::width_checker<fmt::v6::internal::error_handler>, fmt::v6::basic_format_context<std::back_insert_iterator<fmt::v6::internal::buffer<char> >, char> >(fmt::v6::internal::width_checker<fmt::v6::internal::error_handler>&&, fmt::v6::basic_format_arg<fmt::v6::basic_format_context<std::b [... cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [orb_slam3_ros_wrapper/CMakeFiles/orb_slam3_ros_wrapper_mono.dir/build.make:261: /home/odroid/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono] Error 1
make[1]: *** [CMakeFiles/Makefile2:1654: orb_slam3_ros_wrapper/CMakeFiles/orb_slam3_ros_wrapper_mono.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j1" failed
```

but i have original ORB_SLAM3 working on it.
fmt i installed with:
`sudo apt install libfmt-dev`

zoldaten, 2022-11-17:
ok. seems i fix it:
`nano orb_slam3_ros_wrapper/include/common.h `
put there:
`#define FMT_HEADER_ONLY`

built successfully.

thien94, 2022-11-17:
Thanks for letting us know the solution!

closed 2022-11-17 by thien94