thien94/orb_slam3_ros_wrapper #12: OpenCV

RoZhong, 2022-03-08:


RoZhong, 2022-03-08:
Because I am afraid of version problems, I am using ORB_slam3-v0.3. There is no problem in running ORB3 alone, but after adding ros_wrapper, there will be an OpenCV error when the optimization ends. I am using OpenCV3.4.15. The error is as follows:

IMU frequency: 200 Hz
IMU gyro noise: 0.00017 rad/s/sqrt(Hz)
IMU gyro walk: 1.9393e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.002 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.003 m/s^3/sqrt(Hz)
First KF:0; Map init KF:0
New Map created with 473 points
Point distribution in KeyFrame: left->473 --- right->0
[ INFO] [1646701879.260294406]: Finished waiting for tf, waited 13.002301 seconds
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
OpenCV Error: Assertion failed (isIdentity(expr)) in _InputArray, file /home/zyf_/libraries/opencv-3.4.15/modules/core/src/matrix_expressions.cpp, line 1843
terminate called after throwing an instance of 'cv::Exception'
  what(): /home/zyf_/libraries/opencv-3.4.15/modules/core/src/matrix_expressions.cpp:1843: error: (-215) isIdentity(expr) in function _InputArray

[orb_slam3_mono_inertial-2] process has died [pid 8090, exit code -6, cmd /home/zyf_/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial /camera/image_raw:=/cam0/image_raw /imu:=/imu0 __name: =orb_slam3_mono_inertial __log:=/home/zyf_/.ros/log/a0e09940-9e7c-11ec-9a49-ac2b6e3d0ecd/orb_slam3_mono_inertial-2.log].
log file: /home/zyf_/.ros/log/a0e09940-9e7c-11ec-9a49-ac2b6e3d0ecd/orb_slam3_mono_inertial-2*.log

vladimirzhuravlev09, 2022-07-11:
try OpenCV 4.4 or 4.2

thien94, 2022-08-24:
Hi @RoZhong , sorry for the late reply. 
FYI, I have updated the repo to match V1.0 of ORB-SLAM3. If you are still interested, you can test the latest version and let me know if the issue persists.

thien94, 2022-12-06:
Feel free to reopen this issue if you still have any issues.

closed 2022-12-06 by thien94