thien94/orb_slam3_ros_wrapper #14: Build Error

Mothish97, 2022-07-21:
I am getting  "fatal error: sophus/se3.hpp: No such file or directory" when I try to build the package. I am using Noetic with Ubuntu 20. 

 
[image]

Mothish97, 2022-07-21:
sudo apt-get install ros-noetic-sophus

closed 2022-07-21 by Mothish97

dazory, 2023-12-06:
Alternatively, add `${ORB_SLAM3_DIR}/Thirdparty/Sophus` into `include_directories` like this:

```
include_directories(
   ${ORB_SLAM3_DIR}
   ${ORB_SLAM3_DIR}/include
   ${ORB_SLAM3_DIR}/include/CameraModels
   ${ORB_SLAM3_DIR}/Thirdparty/Sophus # <--
   ${PROJECT_SOURCE_DIR}/include
   ${EIGEN3_INCLUDE_DIR}
   ${catkin_INCLUDE_DIRS}
   ${Pangolin_INCLUDE_DIRS}
)
```