thien94/orb_slam3_ros_wrapper #15: Frame Transformation

koksyuen, 2022-07-29:
[image]
Hi, does anyone understand why the transformation is written as shown in the source code? Any formula is followed?

thien94, 2022-08-15:
Hi, sorry for the late reply.

The general formulation is `T_cam_ros = T_ros_orb x T_cam_orb x T_orb_ros`. `T_cam_orb` is the original pose from ORB-SLAM3, `T_cam_ros` is the transformed pose so that it follows ROS convention.

> What do the "on camera coordinate" and "on map coordinate" means?

These comments are a bit outdated, but the map coordinate refers to the ROS' world frame while camera coordinate refers to the camera frame in ORB-SLAM3.

> I thought ORBSLAM3 shall take the estimated initial pose as the map coordinate origin, right?

That is the case without IMU. When there is IMU, the world frame will be aligned with gravity.

thien94, 2022-10-08:
The frame transformation has been removed in the latest version.

closed 2022-10-08 by thien94