thien94/orb_slam3_ros_wrapper #16: Gtk conflicting error

ani-4nirudh, 2022-08-31:
Hello Thien, 
The command `roslaunch orb_slam3_ros_wrapper orb_slam3_euroc_mono.launch` is giving the following output:
```
(ORB-SLAM3: Current Frame:26263): Gtk-ERROR **: 19:39:01.227: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
[orb_slam3_mono-1] process has died [pid 26263, exit code -5, cmd /home/xxx/ros1_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono /camera/image_raw:=/usb_cam/image_raw __name:=orb_slam3_mono __log:=/home/xxx/.ros/log/618a0dba-294d-11ed-acd5-79ec372d828f/orb_slam3_mono-1.log].
log file: /home/xxx/.ros/log/618a0dba-294d-11ed-acd5-79ec372d828f/orb_slam3_mono-1*.log
[ WARN] [1661967556.988548827]: No transform between frames /world and /camera available after 20.002753 seconds of waiting. This warning only prints once.
```
I tried searching for the answers for this gtk conflict, but haven't found anything useful. Another note, the ROS node for [ORBSLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3) viewer is working completely fine. 

Any help would be appreciated. Thanks :)

thien94, 2022-09-01:
Hi @ani-4nirudh , I have not encountered this error before so I am not sure what is the real issue.

Firstly, maybe you can try to install gtk and gtk3 module
`sudo apt install libcanberra-gtk-module libcanberra-gtk3-module`

The issue might be related to opencv, as suggested in [this discussion](https://forums.developer.nvidia.com/t/the-issue-of-gtk-2-and-3-during-running-opencv-on-agx/127118) and [this one](https://itecnote.com/tecnote/c-opencv-error-gtk-2-x-symbols-detected-using-gtk-2-x-and-gtk-3-in-the-same-process-is-not-supported/). Hopefully you can find some helpful pointers from them.

ani-4nirudh, 2022-09-01:
Hey @thien94, thanks for your reply. You were correct it seems, that there are some OpenCV compatibility issues. For the ORBSLAM_3 it is better to use release 4.4.0. 
- Instructions for building OpenCV are [here](https://github.com/Mauhing/ORB_SLAM3/blob/master/README.md) and [here](https://vitux.com/opencv_ubuntu/). The only difference is, that instead of using the command `cmake -D CMAKE_BUILD_TYPE=Release -D WITH_CUDA=OFF -D CMAKE_INSTALL_PREFIX=/usr/local ..`, replace it with `cmake -D CMAKE_BUILD_TYPE=Release -D WITH_CUDA=OFF -D WITH_GTK=ON -D WITH_GTK3=ON -D WITH_GTK_2_X=OFF -DENABLE_PRECOMPILED_HEADERS=OFF -D CMAKE_INSTALL_PREFIX=/usr/local  ..`. Rest of the instructions for OpenCV build can be followed as given in the two sources above. 
- Then, one can go ahead with rebuilding ORBSLAM3 (The bash scripts `./build.sh` & `./build_ros.sh`), and rebuild your ROS repo. Hopefully that removes this error. 

Another useful source, [here](https://www.freebasic.net/forum/viewtopic.php?t=21287).

So, I will go ahead and close this issue. Thanks again for all the help :)

closed 2022-09-01 by ani-4nirudh