thien94/orb_slam3_ros_wrapper #18: Cant run the Euroc examples

duyanhham123, 2022-10-27:
When I run "roslaunch orb_slam3_ros_wrapper euroc_monoimu.launch", this error occurs:

[orb_slam3_mono_inertial-2] process has died [pid 4454, exit code -6, cmd /home/duyanhhamster/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial /camera/image_raw:=/cam0/image_raw /imu:=/imu0 __name:=orb_slam3_mono_inertial __log:=/home/duyanhhamster/.ros/log/e17e5544-5594-11ed-b936-9b2b7a14024b/orb_slam3_mono_inertial-2.log].
log file: /home/duyanhhamster/.ros/log/e17e5544-5594-11ed-b936-9b2b7a14024b/orb_slam3_mono_inertial-2*.log

When the rviz pop up, there is no TF data. Do you know what's the problem with this?

thien94, 2022-10-27:
Hi @duyanhham123 , thank you for testing out the code.

The error message is too general to pinpoint the issue. Can you find more details from the terminal output or the log file (`log file: /home/duyanhhamster/.ros/log/e17e5544-5594-11ed-b936-9b2b7a14024b/orb_slam3_mono_inertial-2*.log`)?
The error could be in the launch file, for example nonexistent settings/voc file, or it could be with ORB-SLAM3 itself (can you run ORB-SLAM3's ROS example without the wrapper?)

 > When the rviz pop up, there is no TF data. Do you know what's the problem with this?

Because the ros node has died, there is nothing publishing to TF. This should be fixed together with the roslaunch issue.

duyanhham123, 2022-10-27:
I'm running into some issue when running ORB_SLAM3 as i don't know what is the PATH_TO_SETTINGS_FILE really mean, since I can't find any setting file on the repository. 
[image]

I use the ORB_SLAM3 repo from this guy: https://github.com/UZ-SLAMLab/ORB_SLAM3, just like your instruction. However, I noticed that this repo was only tested to Ubuntu 16.04 and 18.04, while I'm using 20.04, so I'm not sure if it's an issue or not?

duyanhham123, 2022-10-27:
Now when I run the roslaunch command again, the error is not there anymore, just a warning like this
[image]

I think my main issue should be related to TF data.

thien94, 2022-10-27:
Good to know the issue is resolved.
It seems you are running ORB-SLAM3 on a virtual machine. In that case, you should monitor the CPU and RAM, as sometimes high CPU or RAM usage can cause ROS to crash.

duyanhham123, 2022-10-27:
Sometimes the error still occurs as it can't find the ORBvoc file as its non-existent in the repository.
[image]

thien94, 2022-10-27:
You should copy the `ORBvoc.txt` file from the folder `ORB-SLAM3/Vocabulary/` (located in the original ORB-SLAM3) to the `config` folder (located in this wrapper package). 
Alternatively, you can change the `voc_file` param in the launch file to point to the right location.

duyanhham123, 2022-10-27:
I have copied that folder, everything seems fine, now i got this new error
[image]

And of course the tf2 issue still there, and im reading the tutorials to know more on this.

thien94, 2022-10-27:
I have not encountered this error before unfortunately, so I am not sure how to fix it. Perhaps these discussions could help:
https://answers.ros.org/question/207786/message-size-limit/ 
https://answers.ros.org/question/212103/using-3rd-party-tcp-socket-within-ros-service-fails/

duyanhham123, 2022-10-28:
Hi,
At the moment I'm using a much more powerful computer to run this. As i run **roslaunch orb_slam3_ros_wrapper euroc_monoimu.launch**, i got this error

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
[orb_slam3_mono_inertial-2] process has died [pid 9618, exit code -11, cmd /home/mallab/EE4002D/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial /camera/image_raw:=/cam0/image_raw /imu:=/imu0 __name:=orb_slam3_mono_inertial __log:=/home/mallab/.ros/log/55eb07ae-5682-11ed-8659-f5ca69d3bfbe/orb_slam3_mono_inertial-2.log].
log file: /home/mallab/.ros/log/55eb07ae-5682-11ed-8659-f5ca69d3bfbe/orb_slam3_mono_inertial-2*.log
[ WARN] [1666935351.106459502]: No transform between frames /world and /camera available after 20.002341 seconds of waiting. This warning only prints once.

Is the process died because I don't provide tf2 data to transform between /world and /camera?

thien94, 2022-10-28:
The `tf` data should not crash the ros node in this way, so it shouldn't be the real issue. Based on the exit code -11 it seems a segmentation fault occurred, maybe the ORB-SLAM3. 

Can you check that the param file (`EuRoC.yaml`) is the same as the one in the `ORB_SLAM3\Examples`Monocular-Inertial` folder? Note that all of the folders (Monocular/Stereo/Monocular-Inertial etc.) contain a file with the same name `EuRoC.yaml` but the content is different, so you need to pick the right one from the correct folder.

duyanhham123, 2022-10-28:
Hi,
I'm able to run euroc example now, but i used the launch file for monocular instead of monocular-intertial. So I guess that it works. Do you have any recommendation on how to run this on raspi camera? Just curious.

thien94, 2022-10-28:
Glad it works out!

I have not tried running ORB-SLAM3 with raspi camera before, but without IMU you can find [instructions for ORB-SLAM2+raspi camera](https://courses.ece.cornell.edu/ece6930/ECE6930_Spring16_Final_MEng_Reports/SLAM/Real-time%20ROSberryPi%20SLAM%20Robot.pdf) and everything should still apply, or a specialized package such as https://github.com/jtabor/RaspberryPiOrbSlam2.

duyanhham123, 2022-11-01:
Hi, even though last week the code works with euroc example data, now when i run it again, a brand new error pops up. 

**(ORB-SLAM3: Current Frame:14406): Gtk-ERROR **: 15:38:00.901: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported**
[orb_slam3_mono_inertial-1] process has died [pid 14406, exit code -5, cmd /home/capstone/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono_inertial /camera/image_raw:=/cam0/image_raw /imu:=/imu0 __name:=orb_slam3_mono_inertial __log:=/home/capstone/.ros/log/8df6fdf8-59af-11ed-963a-eb3297c1feb4/orb_slam3_mono_inertial-1.log].
log file: /home/capstone/.ros/log/8df6fdf8-59af-11ed-963a-eb3297c1feb4/orb_slam3_mono_inertial-1*.log
[rviz-2] process has finished cleanly
log file: /home/capstone/.ros/log/8df6fdf8-59af-11ed-963a-eb3297c1feb4/rviz-2*.log
[ WARN] [1667288295.517622687]: No transform between frames /world and /camera available after 20.002960 seconds of waiting. This warning only prints once.
^C[orb_slam3_ros/trajectory_server_orb_slam3-3] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I have no idea what this error in bold is about. 

In addition, i have done research on the "process has died" error and i think the param file (euroc.yaml) must got some issue, band i notice that there is some difference between euroc file on ros_wrapper and orb_slam3. Then which euroc file should I use for both orb-slam3 and ros_wrapper workspace?

thien94, 2022-11-01:
Hi @duyanhham123 , this error can happen when there is a mismatch between GTK version for OpenCV and ORB-SLAM3. However, it is strange that the error suddenly appears if things were working fine. 

Anyway, you can follow the instructions here https://github.com/thien94/orb_slam3_ros_wrapper/issues/16#issuecomment-1234179588. Basically, you need to build OpenCV 4.4 and then rebuild ORB-SLAM3.

> Then which euroc file should I use for both orb-slam3 and ros_wrapper workspace?

For each type of node (for example, `type="orb_slam3_ros_wrapper_mono_inertial"` at the beginning of `euroc_monoimu.launch` file), the corresponding yaml file from ORB-SLAM3 folder needs to be used (so mono node -> Monocular folder, stereo node -> Stereo folder etc.) Otherwise, some parameters will be missing and there will be errors.

However, the file from the Stereo-Inertial folder ([this one](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Stereo-Inertial/EuRoC.yaml)) should be the most general and can be used for any Mono/Stereo±IMU nodes. You should use this one if you are unsure which one to use.

duyanhham123, 2022-11-01:
Hi, the first error has gone (and i hope that i wont come back again) . To deal with the second error, i copy and paste the stereo-inertial euroc.yaml file from orb-slam3 to euroc.yaml file in ros wrapper workspace. but the same error, with different exit code.
[image]

thien94, 2022-11-02:
Hi @duyanhham123, it is strange that the errors on your system change every time. I tested with the stereo-inertial's euroc.yaml file on my setup and the node still runs fine. You can try each of the following ideas individually and see which one works:
- Deactivate pangolin viewer by adding this parameter in the launch file as: `<param name="enable_pangolin" type="bool" value="false"/>`
- Use the yaml file from the monocular-inertial, check that it works, then add the new lines from stereo-inertial (if you want to test the stereo-inertial mode) and pinpoint which new lines create the error.
- If you will mainly use ROS for your application, I suggest checking out [this package](https://github.com/thien94/orb_slam3_ros) where I combine both the wrapper and ORB-SLAM3 into one.

duyanhham123, 2022-11-04:
Hi, apparently it doesn't run all the time, I need several tries to get it to run. Now it's working find now, I will close this issue then.

closed 2022-11-04 by duyanhham123

kneetate, 2023-02-15:
Hi, I occurred same problems, but fixed. It maybe OpenCV's problems. In my case, errors occurred when cv::resize() called in the function ORBextractor::ComputePyramid(cv::Mat image) in Line 1183, ORBextractor.cc.
I changed opencv 4.7 to opencv  4.2 to fix it. Bugs in resize were reported in the follow link. 
https://stackoverflow.com/questions/69622409/resizing-image-leads-to-segmentation-fault-opencv-c
I hope it can be of any helps to someone.