thien94/orb_slam3_ros_wrapper #19: Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

ToshikiNakamura0412, 2022-10-28:
When I run "roslaunch orb_slam3_ros_wrapper euroc_monoimu.launch", this error occurs:
[image]

The above error occurs when map viewer is started. (Map viewer is not displayed.)
Do you know what's the problem with this?

I can run UZ-SLAMLab/ORB_SLAM3 v1.0-release. (EuRoC Examples and ROS Examples)

Environment
- Ubuntu 20.04
- ROS Noetic 
- OpenCV 4.4.0

thien94, 2022-10-28:
Hi, perhaps one of the libraries uses gtk 2.0 instead of 3.0 which creates an incompatibility issue (the error)
https://github.com/raulmur/ORB_SLAM2/issues/404  https://github.com/raulmur/ORB_SLAM2/issues/325.
- First, check that gtk3.0 is installed with `sudo apt-get install libgtk-3-dev`
- Then, try recompile ORB-SLAM3 from scratch then recompile the ros wrapper.
Hope this helps.

ToshikiNakamura0412, 2022-10-28:
I did the following
- Install libgtk-3-dev
- Delete the build directory and recompile ORB-SLAM3
- Delete catkin build related files and recompile row wrapper

But, same error occurs.

[image]
↓
[image]

thien94, 2022-10-28:
Most likely it is the same issue as https://github.com/thien94/orb_slam3_ros_wrapper/issues/16. Can you try following the same steps to install OpenCV 4.4.0?

ToshikiNakamura0412, 2022-10-28:
I also installed according to [Mauhing/ORB_SLAM3/README.md](https://github.com/Mauhing/ORB_SLAM3/blob/master/README.md). 
I changed the cmake options when building OpenCV as #16. Then I recompile ORB-SLAM3 and ros wrapper.
The problem was solved.
Thank you very much for your advice.

closed 2022-10-28 by ToshikiNakamura0412