thien94/orb_slam3_ros_wrapper #20: WAITING FOR IMAGES Black Screen

9Volts2Ground, 2022-11-11:
I have live camera data coming out of the image_transport `republish`node that is being piped into the orb_slam3_ros_wrapper mono node. You can see the Node Graph below showing the image being piped in. 

[image]

Throwing print statements into the mono node code shows that the subscriber is indeed grabbing camera data. ORB_SLAM3 appears to open up properly. However, I get the sad black screen showing "WAITING FOR IMAGES", as seen below.

[image]

Any idea what is preventing ORB_SLAM3 from working, when camera data is clearly being fed into it? Thanks for the help!

thien94, 2022-11-11:
Hi @9Volts2Ground , thanks for testing out the package.

Assuming that the camera data is actually coming in (by viewing the image with `rqt_image_view`), the monocular mode of ORB-SLAM3 will not start working until there are some initial movements. So move the camera around a bit and it should run normally.

9Volts2Ground, 2022-11-11:
Thanks for the very speedy reply. You're absolutely right, I just had to start moving the camera around and it worked right away. Thank you!

This may be a little off topic, but I am using this package for a project. Would you be open to me submitting a PR with some recommendations based on my findings and experience with the package in the next month or so?

thien94, 2022-11-12:
It's great that this package can be helpful to you. Hope your project works out well.

> Would you be open to me submitting a PR with some recommendations based on my findings and experience with the package in the next month or so?

Absolutely yes. Feel free to submit a PR. We can then discuss further depending on the nature of your recommendations.

In principle, I would refrain from anything that fundamentally changes the underlying algorithm as that might make a wrong impression for ORB-SLAM3. Stuffs like that should be submitted directly to the original ORB-SLAM3.
Otherwise, things like extra functions/improvements/interfaces/improving the readability of the codebase are most welcome.

closed 2022-11-13 by thien94