thien94/orb_slam3_ros_wrapper #22: [orb_slam3_mono-1] process has died exit code -6

MagnusErler, 2022-12-03:
I'm getting the following error when using `roslaunch orb_slam3_ros_wrapper euroc_mono.launch`

```
[orb_slam3_mono-1] process has died [pid 59511, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/orb_slam3_ros_wrapper/orb_slam3_ros_wrapper_mono __name:=orb_slam3_mono __log:=/home/ubuntu/.ros/log/e093e46c-7305-11ed-8581-558ecd5b9b83/orb_slam3_mono-1.log].
log file: /home/ubuntu/.ros/log/e093e46c-7305-11ed-8581-558ecd5b9b83/orb_slam3_mono-1*.log
```

RVIZ shows up and I can see my video feed. I'm also able to see the SLAM working for just one frame before the ORB-SLAM crashes and RVIZ is left behind.

The log seems not be available...

thien94, 2022-12-06:
Hi @MagnusErler , it seems the error is on OpenCV's side instead of the wrapper. Maybe you can check this out https://stackoverflow.com/a/33819978

Gigerin, 2023-04-03:
> Hi @MagnusErler , it seems the error is on OpenCV's side instead of the wrapper. Maybe you can check this out https://stackoverflow.com/a/33819978

I also have the same issue. i checked out you link and looked for  cv.VideoCapture(0) line in all files with grep, found nothing meaningful. Do you have any other ideas?