thien94/orb_slam3_ros_wrapper #23: How to suppress pangolin and openCV image window output

Mechazo11, 2022-12-05:
Hi,

Thank you very much for packaging the ORB-SLAM3 code within the ROS ecosystem supporting ROS Noetic. 

I am using ORB-SLAM3 as the base code for a project that I am working on. 

Which file(s) do I change so that I can suppress the pangolin output and opencv image window? At present, I am not interested in seeing visuals just the output of various states that I push to the console

thien94, 2022-12-06:
HI @Mechazo11 , thank you for your interest in this project.

You can define the following parameter in the launch file to deactivate the Viewer (Pangolin-related only)
`<param name="enable_pangolin" type="bool" value="false"/>`

Mechazo11, 2022-12-07:
@thien94 Thank you for the quick feedback. I will try out your suggestion and get back to you by tomorrow before I close this thread

Mechazo11, 2022-12-12:
Hi @thien94, your suggestion worked. Sorry for the late response. I noticed that while suppressing the pangolin window, the parameter is also suppressing the OpenCV window which shows the detected keypoints in green. Can u direct me to the file where the logic to show and suppress pangolin (and Opencv output) is written?

closed 2022-12-12 by Mechazo11