thien94/orb_slam3_ros_wrapper #26: where the map saved ?

zoldaten, 2022-12-22:
hello.
i put to EuRoC.yaml the line to save a map:
System.SaveAtlasToFile: "~/home/pi/my_map1"
but i seems i doesnt working.
May be ros_wrapper saves map and trajectory somewhere else ?

thien94, 2022-12-23:
Hi @zoldaten , ORB-SLAM3 uses the param `System.SaveAtlasToFile` to specify only the name of the file, while the directory to be saved in is the `ROS_HOME` folder (which is `~/.ros` by default).

zoldaten, 2022-12-23:
thanks for the answer. but cant see anything there - only caches and logs.
 may be i should walk around to make a better map and then press Stop button at GUI ORB_SLAM3 as i do ?
[image]

thien94, 2022-12-23:
> may be i should walk around to make a better map and then press Stop button at GUI ORB_SLAM3 as i do ?

That's a good idea. For more details, ORB-SLAM3 specifies the path file in this function 
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/System.cc#L1403-L1443
You can change `pathSaveFileName` to a specific location if necessary.

zoldaten, 2022-12-23:
ok. got it working. the problem was in node. i used usb_cam node that also work correct. now i move to raspicam-node that lets save a map.
strange effect but work -
[image]
thanks!

closed 2022-12-23 by zoldaten