thien94/orb_slam3_ros_wrapper #28: Issue when running orb_slam3

duyanhham123, 2023-02-14:
Hi,

I'm currently attempting to run the repo for Raspi Camera. I noticed that at the beginning it can record map points, but when I turn the camera to another direction, it just stop recording ( You can take a look at the video attached). Is it normal for your side to run the repo as well?

https://youtu.be/gn2UZD0ZxJ4

thien94, 2023-02-14:
Monocular node of ORB-SLAM3 can lose track easily during rotation-only motion. The reason is that it cannot create new reliable features without some translation motion. See here https://github.com/raulmur/ORB_SLAM2/issues/919 (the monocular node is brought over from ORB-SLAM2 so the same issue applies).

For example, at this timestamp https://youtu.be/gn2UZD0ZxJ4?t=170 onwards, you can see the number of features quickly reduce to zero.

In general, these are the common failure cases for monocular camera: https://github.com/raulmur/ORB_SLAM#7-failure-modes

vigneshwarvenkat, 2023-03-14:
Even when I move the camera forward and make the turn at the same time, the problem persists. I think this is not just happen to rotation-only.

Mechazo11, 2023-03-28:
@vigneshwarvenkat from your video, it may be possible ur system is losing active tracking. Try to see the messages printed out on the terminal window from which you launched your monocular node. If there is a prolonged tracking loss it should say "Track lost" or something similar