thien94/orb_slam3_ros_wrapper #30: No trajectory on RVIZ

vigneshwarvenkat, 2023-02-23:
Our orb_slam3_ros/trajectory topic is not displaying any values in the message (image of the rostopic echo attached) and hence not showing any trajectory path on our RVIZ. Everything else works except for this (i.e. orb_slam3/map_points are showing appropriate values and do appear on the RVIZ as they should. Our hector trajectory package is downloaded. What could be a possible cause and/or solution to this, if you have encountered this?

[image]

thien94, 2023-02-23:
Can you check the param `target_frame_name` and `source_frame_name` in the launch file? By default target_frame_name="/world"  and source_frame_name="/camera", but you need to modify them to match the ones published by ORB-SLAM3.

vigneshwarvenkat, 2023-03-07:
Yes, it works now. Thanks!