thien94/orb_slam3_ros_wrapper #39: rosrun ORB_SLAM3 Mono Couldn't find executable named Mono

MarinHure, 2024-04-10:
Hello, 
I have done all the installations, and succeeded launching ORB-SLAM with the MH_01_easy.bag. 
But when i try to launch 
*rosrun ORB_SLAM3 Mono Vocabulary/ORBvoc.txt Examples/Monocular/EuRoC.yaml* 
I get this error : 
*[rosrun] Couldn't find executable named Mono below /home/user/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3*

Can someone help me resolve this error ?

xytjc, 2024-08-23:
Yeah,I have the same error,have you solved the probelm?

xytjc, 2024-08-23:
Yeah,I have the same error ,have you solve the problem?

build-error, 2025-03-13:
make changes to your `build_ros.sh`
```bash
echo "Building ROS nodes"

cd Examples/ROS/ORB_SLAM3
mkdir build
cd build
cmake .. -DROS_BUILD_TYPE=Release
make -j

cp -v $PROJECTPATH/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/devel/lib/ORB_SLAM3/* $PROJECTPATH/ORB_SLAM3/Examples/ROS/ORB_SLAM
```